Nitrokey/nitrokey-start-firmware #1: multiple definition of `usb_lld_set_data_to_recv'

RaymiiOrg, 2016-08-14:
When trying to compile the firmware the make fails with the text below:

    arm-none-eabi-gcc ../ChibiOS_2.0.8/os/ports/GCC/ARMCMx/crt0.o ../ChibiOS_2.0.8/os/ports/GCC/ARMCMx/STM32F10x/vectors.o  ../ChibiOS_2.0.8/os/ports/GCC/ARMCMx/chcore.o ../ChibiOS_2.0.8/os/ports/GCC/ARMCMx/chcore_v7m.o ../ChibiOS_2.0.8/os/ports/GCC/ARMCMx/nvic.o ../ChibiOS_2.0.8/os/ports/GCC/ARMCMx/cmsis/core_cm3.o ../ChibiOS_2.0.8/os/kernel/src/chsys.o ../ChibiOS_2.0.8/os/kernel/src/chdebug.o ../ChibiOS_2.0.8/os/kernel/src/chlists.o ../ChibiOS_2.0.8/os/kernel/src/chvt.o ../ChibiOS_2.0.8/os/kernel/src/chschd.o ../ChibiOS_2.0.8/os/kernel/src/chthreads.o ../ChibiOS_2.0.8/os/kernel/src/chregistry.o ../ChibiOS_2.0.8/os/kernel/src/chsem.o ../ChibiOS_2.0.8/os/kernel/src/chmtx.o ../ChibiOS_2.0.8/os/kernel/src/chcond.o ../ChibiOS_2.0.8/os/kernel/src/chevents.o ../ChibiOS_2.0.8/os/kernel/src/chmsg.o ../ChibiOS_2.0.8/os/kernel/src/chmboxes.o ../ChibiOS_2.0.8/os/kernel/src/chqueues.o ../ChibiOS_2.0.8/os/kernel/src/chmemcore.o ../ChibiOS_2.0.8/os/kernel/src/chheap.o ../ChibiOS_2.0.8/os/kernel/src/chmempools.o ../ChibiOS_2.0.8/os/hal/src/hal.o ../ChibiOS_2.0.8/os/hal/src/adc.o ../ChibiOS_2.0.8/os/hal/src/can.o ../ChibiOS_2.0.8/os/hal/src/mac.o ../ChibiOS_2.0.8/os/hal/src/pal.o ../ChibiOS_2.0.8/os/hal/src/pwm.o ../ChibiOS_2.0.8/os/hal/src/serial.o ../ChibiOS_2.0.8/os/hal/src/spi.o ../ChibiOS_2.0.8/os/hal/src/mmc_spi.o ../ChibiOS_2.0.8/os/hal/platforms/STM32/hal_lld.o ../ChibiOS_2.0.8/os/hal/platforms/STM32/adc_lld.o ../ChibiOS_2.0.8/os/hal/platforms/STM32/can_lld.o ../ChibiOS_2.0.8/os/hal/platforms/STM32/pal_lld.o ../ChibiOS_2.0.8/os/hal/platforms/STM32/pwm_lld.o ../ChibiOS_2.0.8/os/hal/platforms/STM32/serial_lld.o ../ChibiOS_2.0.8/os/hal/platforms/STM32/spi_lld.o ../ChibiOS_2.0.8/os/hal/platforms/STM32/stm32_dma.o ../boards/NITROKEY_START/board.o ../ChibiOS_2.0.8/os/various/evtimer.o ../ChibiOS_2.0.8/os/various/syscalls.o ../polarssl-0.14.0/library/bignum.o ../polarssl-0.14.0/library/rsa.o ../polarssl-0.14.0/library/aes.o sha256.o call-rsa.o main.o usb_lld.o usb_desc.o usb_ctrl.o usb-icc.o openpgp.o ac.o openpgp-do.o flash.o random.o neug.o sys.o   -mcpu=cortex-m3 -mfix-cortex-m3-ldrd -nostartfiles -Tgnuk.ld -Wl,-Map=./gnuk.map,--cref,--no-warn-mismatch,--gc-sections  -mno-thumb-interwork -mthumb   -o gnuk.elf
    usb_lld.o: In function `usb_lld_set_data_to_recv':
    usb_lld.c:(.text.usb_lld_set_data_to_recv+0x0): multiple definition of `usb_lld_set_data_to_recv'
    main.o:/home/remy/repo/nitrokey-start-firmware/src/usb_lld.h:136: first defined here
    usb_desc.o: In function `usb_lld_set_data_to_recv':
    /home/remy/repo/nitrokey-start-firmware/src/usb_lld.h:136: multiple definition of `usb_lld_set_data_to_recv'
    main.o:/home/remy/repo/nitrokey-start-firmware/src/sys.h:18: first defined here
    usb_ctrl.o: In function `usb_lld_set_data_to_recv':
    /home/remy/repo/nitrokey-start-firmware/src/usb_lld.h:136: multiple definition of `usb_lld_set_data_to_recv'
    main.o:/home/remy/repo/nitrokey-start-firmware/src/sys.h:18: first defined here
    usb-icc.o: In function `usb_lld_set_data_to_recv':
    /home/remy/repo/nitrokey-start-firmware/src/usb_lld.h:136: multiple definition of `usb_lld_set_data_to_recv'
    main.o:/home/remy/repo/nitrokey-start-firmware/src/sys.h:18: first defined here
    sys.o: In function `usb_lld_set_data_to_recv':
    /home/remy/repo/nitrokey-start-firmware/src/usb_lld.h:136: multiple definition of `usb_lld_set_data_to_recv'
    main.o:/home/remy/repo/nitrokey-start-firmware/src/sys.h:18: first defined here
    collect2: error: ld returned 1 exit status
    make: *** [../ChibiOS_2.0.8/os/ports/GCC/ARM/rules.mk:124: gnuk.elf] Error 1

This is the configure line:

    ./configure --target=NITROKEY_START --vidpid="20a0:4211" --enable-keygen --enable-certdo

This is the gcc version:

    gcc version 6.1.1 20160707 (GCC)

RaymiiOrg, 2016-08-14:
On gcc `5.4.0` it does seem to work...

szszszsz, 2016-12-02:
Hi! Sorry for delay.
It is rather compiled with `arm-none-eabi-gcc` than with usual `gcc`. Could you post its version? It is probably `4.9` but I want to make sure.
On GNUK 1.2 `--enable-keygen` is not supported now. Without this one I could successfully compile source with `arm-none-eabi-gcc (15:4.9.3+svn231177-1)`. Could you confirm (branch gnuk1.2)?
If you want I can check original (1.x) firmware regarding this.

szszszsz, 2016-12-02:
Updated comment above

szszszsz, 2016-12-29:
Closing as fixed. Please reopen if the issue still occurs.

closed 2016-12-29 by szszszsz