Nitrokey/nitrokey-start-firmware #2: AttributeError: 'NoneType' object has no attribute 'cmd_verify'

RaymiiOrg, 2016-08-14:
When trying to load a public key for firmware upgrades the following error occurs:

    ./tool/gnuk_put_binary_libusb.py -k 0 CB1522E7.bin
    Traceback (most recent call last):
      File "./tool/gnuk_put_binary_libusb.py", line 112, in <module>
        main(fileid, is_update, data, passwd)
      File "./tool/gnuk_put_binary_libusb.py", line 52, in main
        gnuk.cmd_verify(BY_ADMIN, passwd)
    AttributeError: 'NoneType' object has no attribute 'cmd_verify'

This is on Ubuntu 14.04. GnuPG recognizes the card.

RaymiiOrg, 2016-08-14:
I do have python pyusb 1.0.0 installed as that is in the repo's. 

Reading here:

https://github.com/ggkitsas/gnuk/blob/master/NEWS#L55

    ** Tools and test suite now work with PyUSB 1.0, too.
    It only worked with PyUSB 0.4.3, but it works with PyUSB 1.0 too.

Makes me think this is a version issue.

RaymiiOrg, 2016-08-14:
The gnuk source, tag release/1.2.1, also gives the same error. :(

szszszsz, 2016-12-02:
Hi!
Error might show that GNUK tool does not see any supported device. Possibly one has to add vid/pid of NK Start to the tool's source.

szszszsz, 2016-12-02:
Could you run the same using [gnuk1.2-regnual-fix](https://github.com/Nitrokey/nitrokey-start-firmware/tree/gnuk1.2-regnual-fix) branch? I have added there Nitrokey Start's vid/pid.

szszszsz, 2016-12-29:
Closing as fixed. Please reopen if it still occurs.

closed 2016-12-29 by szszszsz