Nitrokey/nitrokey-start-firmware #10: Cannot import x509 certificate

szszszsz, 2018-07-05:
Importing x509 certificate results in an error.
It is [mentioned](http://www.gniibe.org/memo/development/gnuk/gnuk-token-card-holder-certificate.html), that GNUK accepts certificate only in binary format, but it is not specified which one. DER format was not working. Perhaps the test file itself was invalid - to check.

Example conversion command using `openssl x509`:
```
openssl x509 -in input.der -inform DER -out output.pem -outform PEM
```

Firmware: RTM.6 / GNUK 1.2.10 (latest GNUK).
Source: https://support.nitrokey.com/t/failed-to-write-x509-cert-to-nitrokey-start/1127

jans23, 2018-07-05:
For the OpenPGP Card, a X.509 certificate is just a blob and you could
store any type of data instead. I guess it is the same for Gnuk. The
most common root cause is that the certificate is too large for the
available space.

szszszsz, 2018-07-05:
I see. Tested sizes were from 1600 to 2300 bytes. Perhaps smaller would do.

alex-nitrokey, 2018-08-03:
btw: related to this [OpenSC issue](https://github.com/OpenSC/OpenSC/issues/1282)

alex-nitrokey, 2018-09-17:
How did you test the import? OpenSC seems to work again now. Please see [the OpenSC issue](https://github.com/OpenSC/OpenSC/issues/1282).

szszszsz, 2019-04-19:
To retest

alex-nitrokey, 2019-04-24:
What exactly? Importing certs is working fine in OpenSC now. The gnuk_put_binary script seems to be broken.

szszszsz, 2019-04-25:
I mean to reproduce the original issue from the forum, as far as possible. If you have confirmed it works earlier, then it could be closed.

szszszsz, 2019-04-25:
Is it the same story, as with https://support.nitrokey.com/t/unable-to-store-signed-certificate-on-nitrokey-start/971 ?

alex-nitrokey, 2019-05-10:
It is indeed still not working for GnuPG. But I have no idea if it ever did in the first place. As far as I know the firmware of the start is a bit special regarding the import of certs, thus the import script (that is currently broken) and special handling in OpenSC for Gnuk (which works fine for OpenSC 0.19).

I think this is not an issue of the firmware. Therefore, I would close this issue here. We could ask on gnuk-users to make sure that this is not fixable in GnuPG. As NIIBE is a maintainer of GnuPG I would be surprised if he would not have thought about implementing it if feasible.