Nitrokey/nitrokey-start-firmware #11: tests fail

ib54003, 2018-09-03:
Hey,

i wanted to test my Nitrokey Start after a _factory-reset_ but i get the following error:
```

test_005_personalize_admin_less.py::test_setup_pw3_1 FAILED   [ 36%]

============================= FAILURES ==============================
_________________________ test_setup_pw3_1 __________________________

card = <openpgp_card.OpenPGP_Card object at 0x7f677d22a940>

    def test_setup_pw3_1(card):
>       r = card.change_passwd(3, PW1_TEST2, PW3_TEST1)

test_005_personalize_admin_less.py:273: 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
openpgp_card.py:103: in change_passwd
    return self.cmd_change_reference_data(who, passwd_old + passwd_new)
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _

self = <openpgp_card.OpenPGP_Card object at 0x7f677d22a940>, who = 3
data = b'new user pass phraseanother admin pass phrase'

    def cmd_change_reference_data(self, who, data):
        cmd_data = iso7816_compose(0x24, 0x00, 0x80+who, data)
        sw = self.__reader.send_cmd(cmd_data)
        if len(sw) != 2:
            raise ValueError(sw)
        if not (sw[0] == 0x90 and sw[1] == 0x00):
>           raise ValueError("%02x%02x" % (sw[0], sw[1]))
E           ValueError: 6982

openpgp_card.py:237: ValueError
=============== 1 failed, 149 passed in 15.64 seconds ===============
```

Any help?

alex-nitrokey, 2018-09-03:
What firmware version are you using? For firmware prior to 1.2.6 factory-reset may block the device for ever (except if you do not like to break the device, add a programmer and use the hardware interface :-/).

Please paste the output of ```gpg --card-status``` (at least the reader information), so that we can determine the used firmware.

ib54003, 2018-09-03:
Hey,
sadly its on firmware 1.2.6 so i probably bricked the device with the factory-reset right ?

alex-nitrokey, 2018-09-03:
No, afaik it is fine from 1.2.6 on. (sorry for confusion)

alex-nitrokey, 2018-09-03:
iirc you can not upgrade if the device is bricked.

Do a ```gpg --card-edit``` and have a look at "PIN retry counter". If it is "3 0 3" or "3 3 3", type in ```verify``` put in the User PIN and have look if the retry counter is changing. If not - you are good and the problem is somewhere else. If one of the counters turned zero, the device is probably bricked.

btw: did you had any problems other than that the test failed? For now I don't see what's the problem anyway. Maybe I shot in the wrong direction :thinking:

nicorikken, 2020-04-05:
I just ran into this exact same failed test, on Debian Bullseye (Testing). The key seems to work just fine, but the test fails at this exact same point. The tests are not set-up to handle tear-down of continue other tests to get an idea of the total scope of failures.
As I lack another way of verifying the integrity, I just keep my fingers crossed :crossed_fingers: and not use my Nitrokey Start for real production usage :disappointed:

alex-nitrokey, 2020-04-06:
The tests are not meant for integrity testing, but check if everything is working fine e.g. after adding new features or whatsoever. This is not for firmware verification or to check if your device is faulty etc. Having tests failing does not necessarily mean any harm. They are supposed to fail in various situations. So no worries. Please open a thread in the [forum](https://support.nitrokey.com/), if there is something wrong with the device.

szszszsz, 2020-04-07:
@nicorikken Hi! There are 2 test suites, and one could be not updated. Could you post more details - how the tests were run, what failed, and with what version of the firmware? This way we could check it.