Nitrokey/nitrokey-start-firmware #15: Self flash-read protection

szszszsz, 2019-04-25:
Enable flash read-protection in the firmware, right on the very first device initialization.

Related: #14

coelner, 2020-10-04:
As reminder: https://blog.zapb.de/stm32f1-exceptional-failure/

szszszsz, 2020-10-05:
@coelner Indeed, in this view such solution is not that much needed to implement anymore.

coelner, 2020-10-05:
Or, like the solokey, offer a normal and a hacker version The normal version could be sealed with your glitter epoxy. Without the photo, but atleast used as visible seal?

szszszsz, 2020-10-05:
We thought about that, but epoxy does not scale well for the production. We plan to mitigate this problem by moving to another MCU in the next hardware revision.
As for the user data, these are secure as long as the PIN is not brute-forced, which is why we ask to provide longer PINs since latest firmwares.