Nitrokey/nitrokey-start-firmware #21: Update RTM fails with "Could not connect to the device"

peterroth, 2019-12-11:
Hi everybody,

I followed the steps from prebuilt/README.md to update my Nitrokey Start (to RTM.7), but it fails every time, doesn't matter what I change or do. I have pyusb and pytest:
```
pip3 list | egrep "usb|test"
pytest             5.3.1
pyusb              1.0.2
```
I modified the gnuk_get_random.py, upgrade_by_passwd.py and usb_strings.py to point to the location where my Python 3 is installed.
The error I can see is:
```
*** Could not connect to the device. Attempting to close scdaemon.
*** Running: gpg-connect-agent "SCD KILLSCD" "SCD BYE" /bye
*** Please try again...
*** Could not proceed with the update. Please close other applications, that possibly use it (e.g. scdaemon, pcscd) and try again.
```
I checked and no scdaemon, nor pcscd is running:
```
ps -ef | grep -i scd
  504 44977 36923   0  4:00PM ttys002    0:00.01 grep --colour=always -i scd
```

I read that I should remove the device from the USB plug, reinsert it and run the same command, but it still fails with the same error message.
Oddly, the 'gpg --card-status' doesn't show any RTM version:
```
gpg --card-status
Reader ...........: Nitrokey Nitrokey Start
```
The 'usb_strings.py' shows the device is on RTM.6. My device is the red blinking one.

Do you have any idea what else can be checked or set to overcome this cumbersome situation?

szszszsz, 2019-12-11:
Hi!

What is your attempt counter on the device? It should be listed on `gpg2 --card-status`.

szszszsz, 2019-12-11:
Please make another run with this updated version [1], where the error strings are printed to console.

[1] https://github.com/Nitrokey/nitrokey-start-firmware/blob/21-update-verbose/tool/upgrade_by_passwd.py

peterroth, 2019-12-11:
Thanks, I checked and the counters are on 3:
```
gpg2 --card-status
Key attributes ...: rsa2048 rsa2048 rsa2048
Max. PIN lengths .: 127 127 127
PIN retry counter : 3 3 3
```
I'm not sure if printing the error did the trick (checked the last commit) but with the updated Python script it succeeded this time:
```
Device:
    Vendor: Nitrokey
   Product: Nitrokey Start
    Serial: FS***3
  Revision: RTM.7
    Config: NITROKEY_START:dfu=no:debug=no:pinpad=no:certdo=yes:factory_reset=yes
       Sys: 3.0
```

closed 2019-12-11 by peterroth

szszszsz, 2019-12-12:
I wish we would better pinpoint the cause, but anyway I am glad it works for you!

I have modified the update tool, so it would print the error message right away, so it should be more obvious in the future.

peterroth, 2019-12-16:
I tried on the machine where I couldn't update originally on last Wednesday, when I opened the report; there it still fails with the below error:

> *** Could not connect to the device. Attempting to close scdaemon.
> *** Running: gpg-connect-agent "SCD KILLSCD" "SCD BYE" /bye
> *** Please try again...
> *** Could not proceed with the update. Please close other applications, that possibly use it (e.g. scdaemon, pcscd) and try again.

It is printed after I downloaded the newest available version with 'git clone'.

0rb677, 2020-02-16:
@szszszsz hi. same error, i use master branch

```
(env3) user@nitrokey:~/nitrokey-start-firmware/tool$ sudo apt install python3-usb
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following NEW packages will be installed:
  python3-usb
0 upgraded, 1 newly installed, 0 to remove and 7 not upgraded.
Need to get 38.8 kB of archives.
After this operation, 244 kB of additional disk space will be used.
Get:1 https://deb.debian.org/debian buster/main amd64 python3-usb all 1.0.2-1 [38.8 kB]
Fetched 38.8 kB in 0s (89.8 kB/s)      
Selecting previously unselected package python3-usb.
(Reading database ... 134813 files and directories currently installed.)
Preparing to unpack .../python3-usb_1.0.2-1_all.deb ...
Unpacking python3-usb (1.0.2-1) ...
Setting up python3-usb (1.0.2-1) ...

(env3) user@nitrokey:~/nitrokey-start-firmware/tool$ 
(env3) user@nitrokey:~/nitrokey-start-firmware/tool$ 
(env3) user@nitrokey:~/nitrokey-start-firmware/tool$ sudo apt install python3-cffi
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following additional packages will be installed:
  python3-ply python3-pycparser
Suggested packages:
  python3-dev python-ply-doc
The following NEW packages will be installed:
  python3-cffi python3-ply python3-pycparser
0 upgraded, 3 newly installed, 0 to remove and 7 not upgraded.
Need to get 224 kB of archives.
After this operation, 1,223 kB of additional disk space will be used.
Do you want to continue? [Y/n] y
Get:1 https://deb.debian.org/debian buster/main amd64 python3-ply all 3.11-3 [65.1 kB]
Get:2 https://deb.debian.org/debian buster/main amd64 python3-pycparser all 2.19-1 [75.6 kB]
Get:3 https://deb.debian.org/debian buster/main amd64 python3-cffi all 1.12.2-1 [83.3 kB]
Fetched 224 kB in 0s (568 kB/s)    
Selecting previously unselected package python3-ply.
(Reading database ... 134836 files and directories currently installed.)
Preparing to unpack .../python3-ply_3.11-3_all.deb ...
Unpacking python3-ply (3.11-3) ...
Selecting previously unselected package python3-pycparser.
Preparing to unpack .../python3-pycparser_2.19-1_all.deb ...
Unpacking python3-pycparser (2.19-1) ...
Selecting previously unselected package python3-cffi.
Preparing to unpack .../python3-cffi_1.12.2-1_all.deb ...
Unpacking python3-cffi (1.12.2-1) ...
Setting up python3-ply (3.11-3) ...
Setting up python3-pycparser (2.19-1) ...
Setting up python3-cffi (1.12.2-1) ...
Processing triggers for man-db (2.8.5-2) ...

(env3) user@nitrokey:~/nitrokey-start-firmware/tool$ 
(env3) user@nitrokey:~/nitrokey-start-firmware/tool$ ./upgrade_by_passwd.py -f ../prebuilt/RTM.7/{regnual.bin,gnuk.bin} 
../prebuilt/RTM.7/regnual.bin: 4504
../prebuilt/RTM.7/gnuk.bin: 115712
Currently connected device strings:
Device: 
CRC32: ae1cae42

*** Could not connect to the device. Attempting to close scdaemon.
*** Running: gpg-connect-agent "SCD KILLSCD" "SCD BYE" /bye
*** Please try again...
*** Could not proceed with the update. Please close other applications, that possibly use it (e.g. scdaemon, pcscd) and try again.
(env3) user@nitrokey:~/nitrokey-start-firmware/tool$ sudo apt install scdaemon
Reading package lists... Done
Building dependency tree       
[... 60 lines cut ...]

*** Could not connect to the device. Attempting to close scdaemon.
*** Running: gpg-connect-agent "SCD KILLSCD" "SCD BYE" /bye
*** Please try again...
*** Could not proceed with the update. Please close other applications, that possibly use it (e.g. scdaemon, pcscd) and try again.
(env3) user@nitrokey:~/nitrokey-start-firmware/tool$ sudo ./upgrade_by_passwd.py -f ../prebuilt/RTM.7/{regnual.bin,gnuk.bin} 
../prebuilt/RTM.7/regnual.bin: 4504
../prebuilt/RTM.7/gnuk.bin: 115712
Currently connected device strings:
Device: 
    Vendor: Nitrokey
   Product: Nitrokey Start
    Serial: FSIJ-1.2.6-67092242
  Revision: RTM.5-modified
    Config: NITROKEY_START:dfu=no:debug=no:pinpad=no:certdo=yes:factory_reset=yes
       Sys: 3.0
CRC32: ae1cae42

*** Could not connect to the device. Attempting to close scdaemon.
*** Running: gpg-connect-agent "SCD KILLSCD" "SCD BYE" /bye
*** Please try again...
*** Could not proceed with the update. Please close other applications, that possibly use it (e.g. scdaemon, pcscd) and try again.
(env3) user@nitrokey:~/nitrokey-start-firmware/tool$ 

```