Nitrokey/nitrokey-start-firmware #22: PGP and S/MIME simultanous use

szszszsz, 2019-12-28:
Tracking for https://github.com/Nitrokey/nitrokey-pro-firmware/issues/68

alex-nitrokey, 2020-01-02:
The gnuk/NK Start firmware does not implement OpenPGP Card stuff > v2.x besides ECC keys, afaik. Thus, this would probably need some major changes. If this is done, it would be a good idea to implement all of newer stuff of the spec to make it consistent with "real" cards.

szszszsz, 2020-01-02:
Or at least partially to support this one feature. 
Alternatively, what I had initially in mind for a quick solution is to have a switchable banks with the keys, though it would make some complications, like smart card host-side cache, or how would the banks be selected, hence this seems to be inferior solution.

szszszsz, 2020-04-04:
I have created a ticket for the solution as mentioned in the previous comment.
@alex-nitrokey Would #27 help with this issue in your opinion, until actual update to OpenPGP v3.x features happens?