Nitrokey/nitrokey-start-firmware #23: Please try again, and make sure the entered password is correct.

0rb677, 2020-02-16:
@szszszsz hi please can you help me?
```
user@nitrokey:~/nitrokey-start-firmware/tool$ pip3 list | egrep "usb|test"
pytest             5.3.5  
pyusb              1.0.2  
```
```
 user@nitrokey:~/nitrokey-start-firmware/tool$ ps -ef | grep -i scd
root      3373     1  0 16:45 ?        00:00:00 /usr/sbin/pcscd --foreground --auto-exit
user      3428  2882  0 16:46 ?        00:00:00 scdaemon --multi-server
user      3553  1004  0 16:47 pts/0    00:00:00 grep -i scd
```
```
user@nitrokey:~/nitrokey-start-firmware/tool$ sudo kill -9 3373
user@nitrokey:~/nitrokey-start-firmware/tool$ sudo kill -9 3428
```
```
user@nitrokey:~/nitrokey-start-firmware/tool$ ps -ef | grep -i scd
user      3561  1004  0 16:48 pts/0    00:00:00 grep -i scd
```

```
user@nitrokey:~/nitrokey-start-firmware/tool$ sudo ./upgrade_by_passwd.py -f ../prebuilt/RTM.7/{regnual.bin,gnuk.bin} 
../prebuilt/RTM.7/regnual.bin: 4504
../prebuilt/RTM.7/gnuk.bin: 115712
Currently connected device strings:
Device: 
    Vendor: Nitrokey
   Product: Nitrokey Start
    Serial: FSIJ-1.2.6-67092242
  Revision: RTM.5-modified
    Config: NITROKEY_START:dfu=no:debug=no:pinpad=no:certdo=yes:factory_reset=yes
       Sys: 3.0
CRC32: ae1cae42

Device: 
Configuration: 1
Interface: 0
*** Could not proceed with the update. Please try again, and make sure the entered password is correct.
*** Found error: ('cmd_write_binary 1', '6581')
*** Could not proceed with the update. Please close other applications, that possibly use it (e.g. scdaemon, pcscd) and try again.

```

szszszsz, 2020-02-17:
Hi!

From the log I see, that the communication has worked (so `pcscd` was not obstructing it this time), and the device has returned the `6581` error on binary object write command.
It seems that error is caused by the already used `0` slot, which is selected by default. These slots are writable only once per factory-reset cycle. Potentially you could try to run the update with `-k 1`, but please wait until I will test that first on my hardware. I plan to check it this Wednesday.

Quick questions:
- Have you used any other upgrade script? e.g. `gnuk_upgrade.py`?
- Could you provide the Git revision? (`git describe`)

References:
- https://alioth-lists-archive.debian.net/pipermail/gnuk-users/2016q3/000371.html
- https://alioth-lists-archive.debian.net/pipermail/gnuk-users/2016q3/000372.html
- https://alioth-lists-archive.debian.net/pipermail/gnuk-users/2016q3/000374.html
- https://support.nitrokey.com/t/failed-to-write-x509-cert-to-nitrokey-start/1127/3
- https://lists.gnupg.org/pipermail/gnuk-users/2018-September.txt
- https://lists.gnupg.org/pipermail/gnuk-users/2018-April.txt

szszszsz, 2020-02-17:
- [ ] Make upgrade tool more informative in such case like this one;
- [ ] Propose adding `-k 1` parameter for another run; Try it automatically?
- [ ] Warn about potential on-device data lost.

0rb677, 2020-02-20:
@szszszsz I did everything right according to the instructions. straightforward. from here
https://github.com/Nitrokey/nitrokey-start-firmware/blob/gnuk1.2-regnual-fix/prebuilt/README.md
```
[user@qb1 nitrokey-start-firmware]$ git describe
RTM.7-15-gf278714
```
It works great with Heads firmware http://osresearch.net/. But i can store only one gpg key and use 2048 bits encryption. 4096 not working(?)
Will the firmware update give me any benefits? Sorry my English :)

szszszsz, 2020-02-21:
I see, thank you. Sorry for the delay in the reproduction, will do so in the near time, as this case has to be solved for the future.

The changes are listed here:
- https://github.com/Nitrokey/nitrokey-start-firmware/blob/gnuk1.2-regnual-fix/NEWS
RTM.5 contain Gnuk 1.2.6. RTM.7 is Gnuk 1.2.14.

As far as I am reading it, from the bigger things, the KDF-DO support was added. If you do not plan to use it, and you do not experience any stability issues, you can omit it for now.

For 4096 key, it has to be imported to the device as far as I remember, as its memory does not allow to generate it.

szszszsz, 2024-01-12:
> This token has one identity provisioned, but as far as I understand, resetting to factory defaults is not a prerequisite to an update, right?

Hi @alt3r-3go ! 
AFAIR factory reset is required

szszszsz, 2024-01-16:
Definitely! Feel free to make PRs whenever you see its worth it. 
Perhaps first it would be good to make a ticket with the problem you would like to solve, and ask pynitrokey team whether they have time to look into testing this, if the change you need is more complex. 
I myself got a little busy and can't make promises about checking updates to the Nitrokey Start related routines, but I can imagine that if PR changes are simple (like UI messages corrections), then it should not be kept long before merging.