Nitrokey/nitrokey-start-firmware #27: Idea: multiple identities

szszszsz, 2020-04-04:
Provide a way to simulate multiple OpenPGP 2.x cards on the device.
Could be a workaround for #22.

cc @alex-nitrokey

szszszsz, 2020-04-04:
Mentioned commit triples the user storage, leaving 3kB for firmware changes.

alex-nitrokey, 2020-04-06:
Mh, I would say the best approach would be to support and implement the MANAGE SECURITY ENVIRONMENT (MSE) feature of OpenPGP Card v3.3 specification. This way, OpenSC would just work as in *real* OpenPGP Cards. I am not sure if this approach would work in OpenSC, but I guess it could get really messy.

In the end it can get more and more difficult to get Gnu/Start devices supported by software if they do not stick to the specification imho. So best case would be to support *all* changes of OpenPGP Card v3.3 or none.

Having admin-less PIN usage and ECC independent of any specification has already caused some problems, although these are desirable features.

szszszsz, 2020-05-19:
Merged to main branch with #33. To be tested.

closed 2020-05-27 by szszszsz