Nitrokey/nitrokey-start-firmware #28: Flashing NK Start to 1.2.12 failed

opelx, 2020-04-15:
Hallo,
this is my first update to a newly NK Start, nothing was stored before. I did follow [Firmware upgrade instructions](https://github.com/Nitrokey/nitrokey-start-firmware/tree/gnuk1.2-regnual-fix/prebuilt#firmware-upgrade-instructions).
Before is:

        $ cat before.status 
        Reader ...........: Nitrokey Nitrokey Start (FSIJ-1.2.10-XXXXXXX) 00 00
        Application ID ...: D276000124010200FFFEXXXXXXX0000
        Application type .: OpenPGP
        Version ..........: 2.0
        Manufacturer .....: unmanaged S/N range
        Serial number ....: XXXXXXX
        Name of cardholder: [nicht gesetzt]
        Language prefs ...: [nicht gesetzt]
        Salutation .......: 
        URL of public key : [nicht gesetzt]
        Login data .......: [nicht gesetzt]
        Signature PIN ....: zwingend
        Key attributes ...: rsa2048 rsa2048 rsa2048
        Max. PIN lengths .: 127 127 127
        PIN retry counter : 3 3 3
        Signature counter : 0
        KDF setting ......: off
        Signature key ....: [none]
        Encryption key....: [none]
        Authentication key: [none]
        General key info..: [none]

and the diff gives:

        $ diff before.status after.status
        1c1
        < Reader ...........: Nitrokey Nitrokey Start (FSIJ-1.2.10-XXXXXXX) 00 00
        ---
        > Reader ...........: 20A0:4211:FSIJ-1.2.6-XXXXXXX:0
        17d16
        < KDF setting ......: off

The Reader ID is completely smashed. 

I've got during flashing:

        $ ./upgrade_by_passwd.py ../prebuilt/$RTM/regnual.bin ../prebuilt/$RTM/gnuk.bin
        Admin password: 
        ../prebuilt/RTM.5/regnual.bin: 4372
        ../prebuilt/RTM.5/gnuk.bin: 113664
        Currently connected device strings:
        Device: 
            Vendor: Nitrokey
           Product: Nitrokey Start
            Serial: FSIJ-1.2.10-XXXXXXX
          Revision: RTM.6
            Config: NITROKEY_START:dfu=no:debug=no:pinpad=no:certdo=yes:factory_reset=yes
               Sys: 3.0
        CRC32: b7020271

        Device: 
        Configuration: 1
        Interface: 0
        ./upgrade_by_passwd.py:205: DeprecationWarning: tostring() is deprecated. Use tobytes() instead.
          main(wait_e, keyno, passwd, data_regnual, data_upgrade[4096:], args.bootloader)
        20002a00:20005000
        *** Running update. Do NOT remove the device from the USB slot, until further notice.
        Downloading flash upgrade program...
        start 20002a00
        end   20003b00
        Run flash upgrade program...
        Waiting for device to appear:
          Wait 1 second...
        Device: 
        08001000:08020000
        Downloading the program
        start 08001000
        end   0801bc00
        failure
        failure
        failure
        failure
        failure
        ... over 300x ...
        failure
        failure
        Protecting device
        Finish flashing
        Resetting device
        Update procedure finished. Device could be removed from USB slot.
        Currently connected device strings (after upgrade):
        Device: 
            Vendor: Nitrokey
           Product: Nitrokey Start
            Serial: FSIJ-1.2.6-XXXXXXX
          Revision: RTM.5-modified
            Config: NITROKEY_START:dfu=no:debug=no:pinpad=no:certdo=yes:factory_reset=yes
               Sys: 3.0

The tests failed also:

        $ pytest -vx test_*
        ImportError while loading conftest '/home/olaf/nitrokey/nitrokey-start-firmware/tests/conftest.py'.
        conftest.py:3: in <module>
            from openpgp_card import OpenPGP_Card
        openpgp_card.py:25: in <module>
            from kdf_calc import kdf_calc
        kdf_calc.py:23: in <module>
            from cffi import FFI
        E   ModuleNotFoundError: No module named 'cffi'

A second try failed with:

        $ ./upgrade_by_passwd.py ../prebuilt/$RTM/regnual.bin ../prebuilt/$RTM/gnuk.bin
        Admin password: 
        ../prebuilt/RTM.5/regnual.bin: 4372
        ../prebuilt/RTM.5/gnuk.bin: 113664
        Currently connected device strings:
        Device: 
            Vendor: Nitrokey
           Product: Nitrokey Start
            Serial: FSIJ-1.2.6-XXXXXXX
          Revision: RTM.5-modified
            Config: NITROKEY_START:dfu=no:debug=no:pinpad=no:certdo=yes:factory_reset=yes
               Sys: 3.0
        CRC32: b7020271

        *** Could not connect to the device. Attempting to close scdaemon.
        *** Running: gpg-connect-agent "SCD KILLSCD" "SCD BYE" /bye
        *** Please try again...
        *** Could not proceed with the update. Please close other applications, that possibly use it (e.g. scdaemon, pcscd) and try again.

So, what happened and how to get the current firmware.
Thanks

szszszsz, 2020-04-16:
Hi!

> Edit: Does it correct, that I've done a firmware downgrade unintentionally?

It looks like it unfortunately. The guide does not mention to check the latest release number, but instead shows the RTM.5 as the current one, which might be misleading. You had RTM.6, and the latest one is RTM.7. Sorry for that, it will be corrected.

In any case, this should not break the device, and still allow to upgrade. The second attempt failure could result from a use conflict from other applications.
The tests are failing because of the missing Python module, not due to device failure. 
Let me run the same update path as you first, so I could test the procedure before executing on your side.

cc @alex-nitrokey

alex-nitrokey, 2020-04-16:
I could reproduce the behavior. Upgrading from RTM.5 to RTM.7 works fine. I think you really just needs to make sure that the device is not locked (you may just reboots). I guess starting the tests has locked its usage. Just try again with RTM.7 please.

alex-nitrokey, 2020-04-16:
btw: I guess finding out why the downgrade to RTM.5 is showing these failures is not really necessary as most users will upgrade - not downgrade :)

closed 2020-04-16 by alex-nitrokey

szszszsz, 2020-04-16:
- [ ] make a quick change in the update guide, so that RTM.7 will be used by default, not RTM.5