Nitrokey/nitrokey-start-firmware #30: Can't set user pin after update to RTM.8

gvintux, 2020-05-06:
Hello, I've updated my nitrokey start to RTM.8
I can change admin pin to 14-digit sequence, but can't do that for user pin.
Here is the result, but gpg --card-edit says that 

`
Error changing the PIN: Conditions of use not satisfied
`

```
Device: 
    Vendor: Nitrokey
   Product: Nitrokey Start
    Serial: FSIJ-1.2.15-67094445
  Revision: RTM.8
    Config: NITROKEY_START:dfu=no:debug=no:pinpad=no:certdo=yes:factory_reset=yes
       Sys: 3.0
```

gvintux, 2020-05-06:
I've found my mistake, I should generate key firstly

closed 2020-05-06 by gvintux

szszszsz, 2020-05-07:
Correct. I just checked and this is present in the documentation on the [Start page](https://www.nitrokey.com/documentation/installation#p:nitrokey-start&os:linux) already.
One thing missing is the new 14-PIN limit. 
@alex-nitrokey @jans23 Could you add that there please?

Perhaps I should refer to this page from releases page too.

alex-nitrokey, 2020-05-28:
I did add the PIN requirements some days ago.