Nitrokey/nitrokey-start-firmware #31: Error: Bad PIN!

gvintux, 2020-05-07:
I'm using it with nitrokey start, I've updated it yesterday to RTM.8 (it was RTM.5)
When I'm trying to encrypt data or use PasswordStore, it asks me for PIN. And it worked yesterday.
## Expected Behavior
PIN is OK, data processed successfully
## Current Behavior
It says "Error: Bad PIN! Take away Security Token now and touch TRY AGAIN", but on PC it works perfectly
## Context
I can't decrypt my password store on my phone
## Your Environment
* Android Version: 8.1
* Device Model: Blackberry BB100 (KeyOne)
* OpenKeychain Version: 5.5
* From Google Play or F-Droid?: Google Play (but I tried F-Droid too)

szszszsz, 2020-05-07:
Hi!
Do I understand correctly that it worked yesterday on the mobile, and stopped the next day?

gvintux, 2020-05-07:
Sorry for incorrect explanation. It stopped working after update to RTM.8

szszszsz, 2020-05-07:
Confirmed same behavior in retests. After reinserting the device after the mentioned message, the cached PIN works, so this could be used as a workaround until the proper fix will be issued for the OpenKeychain. Decryption works as well.
Another problem is that after encrypting with signing to the clipboard, the progress bar is not vanishing, though the operation completes in the background and the result is in the clipboard.
Third one is, that the PIN entry keyboard allows only digits to enter, while all ASCII characters should be handled normally (UTF-8 probably too).

I see you have already relayed the issue to https://github.com/open-keychain/open-keychain/issues/2551 as well. I think this is right - I do not see this as a device issue, so I will leave the ticket here open as tracking to keep us informed.
Please update them with the https://github.com/Nitrokey/nitrokey-start-firmware/issues/31#issuecomment-625159283.

gvintux, 2020-05-07:
Now the key has forgotten my pin. I'm trying to cancel it. GPG tool reports: card error. Seems like firmware update was the gates to hell for my nitrokey. I've tried to update firmware again - it doesn't help

closed 2020-05-07 by gvintux

szszszsz, 2020-05-08:
1. I am sorry, but I do not understand: why are you closing the tickets, while they are valid? Your original case is/was definitely application side problem.
2. Could you elaborate on the firmware update? You should be able to downgrade to the previous firmware version and continue to use it as before the update process.
3. Regarding GnuPG error, could you check is the device generally visible after inserting? Could you take `scdaemon` logs?
To do so, in case you are using Linux, you need add to your `~/.gnupg/scdaemon.conf` file (if not existing, please create) following lines:
```
debug-all
debug-ccid-driver
log-file /home/your_user_directory/.gnupg/scdaemon-nitrokey-start.log
```
Please change the path in log-file parameter to your choosing and restart the scdaemon:
```
sudo killall scdaemon
```
Do not use your  real secrets and PIN, as all communication will be included in the log.
After doing so please try again with the command you are trying to use.
Alternatively one can use https://github.com/Nitrokey/gnupg-docker instead, which sets up log inside a Docker container.