Nitrokey/nitrokey-start-firmware #49: Card Number changes Keys with different gpg versions

Boeserwolff, 2020-08-01:
When loading the card via `gpg --card-status` i get different Card numbers on different machines with different gpg versions:
Ubuntu 18.4
gpg (GnuPG) 2.2.4
libgcrypt 1.8.1
<pre>
Reader ...........: 20A0:4211:FSIJ-1.2.15-431xxxxx:0
Application ID ...: D276000124010200FFFE431022520000
Version ..........: 2.0
Manufacturer .....: unmanaged S/N range
Serial number ....: 431xxxxx
...
Key attributes ...: rsa4096 rsa4096 rsa4096
Max. PIN lengths .: 127 127 127
PIN retry counter : 3 3 3
Signature counter : 0
...
sec>  rsa4096/0xXXXXXXXXXXXXXX  erzeugt: xxxx-xx-xx  verfällt: xxxx-xx-xx
                                  Kartennummer: <b>FFFE 431xxxxx</b>
ssb>  rsa4096/0xXXXXXXXXXXXXXX  erzeugt: xxxx-xx-xx  verfällt: xxxx-xx-xx
                                  Kartennummer: <b>FFFE 431xxxxx</b>
ssb>  rsa4096/0XXXXXXXXXXXXXX  erzeugt: xxxx-xx-xx  verfällt: xxxx-xx-xx
                                  Kartennummer: <b>FFFE 431xxxxx</b>
</pre>

and with gpg (GnuPG) 2.2.19 
libgcrypt 1.8.5 on Ubuntu 20.4 i get:
<pre>
Reader ...........: 20A0:4211:FSIJ-1.2.15-431xxxxx:0
Application ID ...: D276000124010200FFFE431022520000
Application type .: OpenPGP
Version ..........: 2.0
Manufacturer .....: unmanaged S/N range
Serial number ....: 431xxxxx
...
Login data .......: [nicht gesetzt]
Signature PIN ....: nicht zwingend
Key attributes ...: rsa4096 rsa4096 rsa4096
Max. PIN lengths .: 127 127 127
PIN retry counter : 3 3 3
Signature counter : 0
KDF setting ......: off
...
sec>  rsa4096/0xXXXXXXXXXXXXXX  erzeugt: xxxx-xx-xx  verfällt: xxxx-xx-xx
                                  Kartennummer: <b>FFFE 001xxxxx</b>
ssb>  rsa4096/0xXXXXXXXXXXXXXX  erzeugt: xxxx-xx-xx  verfällt: xxxx-xx-xx
                                  Kartennummer: <b>FFFE 001xxxxx</b>
ssb>  rsa4096/0xXXXXXXXXXXXXXX  erzeugt: xxxx-xx-xx  verfällt: xxxx-xx-xx
                                  Kartennummer: <b>FFFE 001xxxxx</b>
</pre>

szszszsz, 2020-08-03:
Hi!
1. Which firmware version do you use? You can check that with [pynitrokey] (command: `nitropy start list`). We have 3 releases with this ID unfortunately (RTM.8 through 10). 
2. In RTM.9 we have zeroed the first digit for the multiple identity feature, which was reverted for the first identity in the RTM.10. Could you check your firmware version and update to RTM.10, if its older?
See these links for instructions:
- https://github.com/Nitrokey/pynitrokey#firmware-update-1
- https://github.com/Nitrokey/nitrokey-start-firmware/releases

[pynitrokey]: https://github.com/Nitrokey/pynitrokey#pynitrokey

Boeserwolff, 2020-08-03:
Hey, 
I use the RTM.10 Firmware already:
```$ nitropy start list
*** Nitrokey tool for Nitrokey FIDO2 & Nitrokey Start
:: 'Nitrokey Start' keys:
FSIJ-1.2.15-43102252: Nitrokey Nitrokey Start (RTM.10)```

szszszsz, 2020-08-03:
I see. I will try to reproduce it locally then.

szszszsz, 2020-08-06:
Hi!
I have looked into this today and could not reproduce. I have noticed in the logs you have published though that the problem lies within the GnuPG's private keys stubs. Just removing them for the `00...` prefixed key and generating again will solve the problem.
Will write more later.