Nitrokey/nitrokey-start-firmware #56: How to use the backup of the secret key?

osresearch, 2020-09-08:
When using the `gpg2 --card-edit` `generate` command, one of the options is to create an off-device backup of the secret.  How do you use this file? It seems that gpg2 hasn't changed since [this mailing list post in 2017](https://lists.gnupg.org/pipermail/gnupg-users/2017-October/059431.html) with regards to importing a secret key to match it to the device.

Some things I've tried:
```
% gpg2 gnupg/sk_4FC1E4C20F8AC356.gpg  
gpg: WARNING: no command supplied.  Trying to guess what you mean ...
gpg: DBG: FIXME: merging secret key blocks is not anymore available
gpg: DBG: FIXME: No way to print secret key packets here
```

```
% gpg2 --allow-secret-key-import --import gnupg/sk_4FC1E4C20F8AC356.gpg  
gpg: key 4FC1E4C20F8AC356: no user ID
gpg: Total number processed: 1
gpg:       secret keys read: 1
```

```
% gpg2 --edit-key 4FC1E4C20F8AC356
gpg (GnuPG) 2.2.19; Copyright (C) 2019 Free Software Foundation, Inc.
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.

pub  rsa2048/E7277F2F7193A538
     created: 2020-09-08  expires: 2021-09-08  usage: SC  
     trust: unknown       validity: unknown
sub  rsa2048/D03ED8B060CD46F0
     created: 2020-09-08  expires: 2021-09-08  usage: A   
sub  rsa2048/4FC1E4C20F8AC356
     created: 2020-09-08  expires: 2021-09-08  usage: E   
[ unknown] (1). TESTING

gpg> toggle
Need the secret key to do this.

gpg> bkuptocard
Need the secret key to do this.
```

jans23, 2020-09-09:
Does [this documentation](https://www.nitrokey.com/documentation/openpgp-create-backup) help?

In general, please post usage-related questions in [our support forum](https://support.nitrokey.com/).

szszszsz, 2020-09-09:
Hi!
With on-device generation only the encryption key's secret (usage: E) can be backed up - I guess you are more interested in the whole key backup, which is linked in https://github.com/Nitrokey/nitrokey-start-firmware/issues/56#issuecomment-689643363 by @jans23 .