Nitrokey/nitrokey-start-firmware #57: ACK button for signatures

fgervais, 2020-09-22:
It seems that ACKBTN support is enabled by default since RTM.7:

https://github.com/Nitrokey/nitrokey-start-firmware/commit/6c721472483cf1b173b4ea92367ef9387ea42448

I have RTM.10 on mine and it's not asking to touch, for signatures at least. Is there something I need to do to enable it?

jans23, 2020-09-22:
There is no physical (touch) button in Nitrokey Start (yet). We will add such physical touch button sooner or later. Potentially this will come with a re-implementation of the firmware. However, no specific plan or roadmap yet. We will tackle this after the upcoming Nitrokey FIDO2 NFC release.

fgervais, 2020-09-22:
Oh but I looked at the kicad files and I see a big touch pad bottom side (pcb botton, key top).

Am I looking at a future revision or something?

fgervais, 2020-09-22:
[image]

jans23, 2020-09-22:
Yes, the hardware layout you are looking at is not what is shipped as NK Start. The correct hardware layout is in Repo nitrokey-pro-hardware.

fgervais, 2020-09-22:
This is the nitrokey start?

[image]

jans23, 2020-09-22:
The hardware repository is [this](https://github.com/Nitrokey/nitrokey-pro-hardware/) and [here](https://github.com/Nitrokey/nitrokey-pro-firmware/) are pictures.

closed 2020-09-22 by jans23