Nitrokey/nitrokey-start-firmware #58: Unable to sign ssh certificate

fgervais, 2020-09-30:
I just found out that I couldn't login with an ssh certificate with my nitrokey start yesterday.

Luckily it seems that this problem as been reported already and fixed on the gnuk side. I actually can't believe my luck, it's been fixed only a couple days ago! What a coincidence.

It seem to be related to this commit: http://git.gniibe.org/cgit/gnuk/gnuk.git/commit/?id=bf30144144498e5ea303942ed5479c9759aa7fb7

You think this could be integrated in the start firmware? I cannot log into my work embedded devices without this.

The more I think about it, it might even have been reported here already as the person in the official bug report states he has a nitrokey start too (https://dev.gnupg.org/T5041). I can't see it in the issue list though.

szszszsz, 2020-09-30:
Hi!
We do not have scheduled next release yet, but we will try to do this asap (next week realistically, hopefully this one).

FYI: @jans23

fgervais, 2020-10-12:
I tried the current `gnuk1.2-regnual-fix` with this patch on top and it doesn't fix my particular problem. I'm just letting you know so you don't rush out a release because of me.

I'm still thinking this is related to my use-case but I it seems I have another issue before getting to this one.

Now that I'm setup to build the firmware I'll see if I can get to the bottom of this.

szszszsz, 2020-10-12:
@fgervais Thank you for the heads up!
It should be released this week nevertheless. Sorry for the delay!

ccxcz, 2021-11-03:
I'm the reporter on the original Gnuk issue and it would indeed be very handy to have this available now.
Any news on the release?

PS: This indeed doesn't affect OP's issue with SSH authentication as that uses different scdaemon method. This is strictly about using Gnuk to sign blobs that are larger than usual hash sizes, such as SSH certificates.