Nitrokey/nitrokey-start-firmware #61: Nitrokey Start - can't change admin PIN

sanosis, 2021-02-23:
Nitrokey Start (old one) updated to latest firmware: FSIJ-1.2.15 (RTM.10)

After updating firmware I can't change Admin PIN, the standard pin 1-8 works, but after entering the new pin I get: "Error changing the PIN: Conditions of use not satisfied".

Factory reset or uploading keys do not change the situation, tested on 0 and 1 identity.

sanosis, 2021-02-23:
Linux Arch, gpg (GnuPG) 2.2.27, libgcrypt 1.9.2-unknown
Tested with openpg card in gemalto reader - changing the pin worked.

szszszsz, 2021-02-23:
Hi!
Since RTM.8 the minimal PIN length without KDF-DO set is 14 characters (both User and Admin). See release note under:
- https://github.com/Nitrokey/nitrokey-start-firmware/releases/tag/RTM.8

When KDF-DO is set, the PIN can be a regular length.
We cannot show this information during setup in GnuPG unfortunately. Sorry for confusion.

closed 2022-05-05 by sanosis