Nitrokey/nitrokey-start-firmware #67: OpenSSH 9.0 support

szszszsz, 2022-05-02:
We have received a report, that Nitrokey Start might have some problems with OpenSSH 9.0 support. 
At the moment we do not know if the cause is located in the Nitrokey Start firmware (RTM.12 / 1.2.15) or the client applications (GnuPG or OpenSSH).

Connected tickets:
- https://dev.gnupg.org/T5931
- https://dev.gnupg.org/T5935
- https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=1008573
- https://bugs.archlinux.org/task/74143

Pasting redacted content of the HT28294 ticket below.

------------------------------

```
I'm currently trying to setup my Nitrokey Start for SSH authentication.

I have setup ~/.gnupg/gpg-agent.conf to use `enable-ssh-support` and
point my `SSH_AUTH_SOCK` to the correct place while ensuring that
`GPG_TTY` is set correctly and that this is used properly in my ssh
config via `Match host * exec "gpg-connect-agent UPDATESTARTUPTTY
/bye"`.
My main key has the [SC] capability (ed25519) and I have two subkeys
(one with [E] (cv25519) and one with [A] (ed25519) capabilities).
You can get it via `sq wkd get dave@******.de`.

I have setup a test host with the exported key `gpg --export-ssh-key
<ID>` and when attempting the ssh connection I get queried for the
nitrokey's PIN.
After entering the pin though I get the infamous:

    sign_and_send_pubkey: signing failed for ED25519 "cardno:FFFE ********" from agent: agent refused operation

and scdaemon provides me with

    2022-05-01 19:25:25 scdaemon[3354468] operation auth result: Conditions of use not satisfied
    2022-05-01 19:25:25 scdaemon[3354468] app_auth failed: Conditions of use not satisfied
    2022-05-01 19:25:25 scdaemon[3354468] DBG: chan_7 -> ERR 100663427 Conditions of use not satisfied <SCD>

(Full log in attachment)

I have also tried adding the [S] capability to the [A] subkey, but this
leads to the same result.

This is gnupg 2.2.35, openssh 9.0p1 and the latest firmware for the
Nitrokey Start (on Arch Linux).

For testing purposes I also provided `-o
KexAlgorithms=-sntrup761x25519-sha512@openssh.com` to ssh to disable the
newer openssh default as there have been issues with it since the
openssh 9 release (these are allegedly fixed with gnupg 2.2.35). This
unfortunately changed nothing.

```

sanosis, 2022-05-03:
Linux Arch, ssh client OpenSSH_9.0p1, OpenSSL 1.1.1n  15 Mar 2022, gpg (GnuPG) 2.2.35, FSIJ-1.2.15-43135739: Nitrokey Nitrokey Start (RTM.10), ed25519

Problems only affect connections to 9p1 servers (maybe 8.9 as weel) - I had no probems with dropbear, 7.5, 8.2 servers…

Disallowing -sntrup761x25519-sha512@openssh.com allows direct connection to 9p1 servers, but does not help with connection behind ProxyJump/-J

sanosis, 2022-05-03:
When used with new full gpg key (3 subkeys, stored on disk) connections work (both direct and ProxyJump) with and without sntrup761x25519-sha512@openssh.com

szszszsz, 2022-05-04:
@sanosis Thank you for sharing! Looks like then the problem lies on the OpenSSH side. Once we will confirm that on our side we can close this ticket.
It would be nice to learn, why it does not work for the original reporter, but this is a case for the Nitrokey's Support Forum.

Leaving here a draft of Docker-based scripts for the further/future versions retesting:
- https://github.com/Nitrokey/nitrokey-3-firmware/tree/ssh-archlinux/ssh/troubles

dvzrv, 2022-05-06:
FWIW: I have the `RTM.12` firmware version on that nitrokey start. Even after setting

```
KexAlgorithms -sntrup761x25519-sha512@openssh.com
```

in `/etc/ssh/ssh_config` I am unable to connect to the target host running openssh 9.0p1 (client running 9.0p1).

I have now also tried with 8.9p1 on the target and client host (still fails).

However, this works with 8.8p1 on the client host and whichever version on the target host (also without the KexAlgorithms modification, as that has been implemented only later to guard against post-quantum attacks AFAIK)!

@sanosis: Does this also work for you with `RTM.12` firmware version?

@szszszsz: What logs can I provide for a better debug scenario?

dvzrv, 2022-05-06:
I sent a mail with ssh and scdaemon logs for openssh 8.8p1 and openssh 9.0p1 to support@nitrokey.com now.

dvzrv, 2022-05-06:
A user in the downstream bug report with Arch Linux noticed that gnuk would have to be >= 1.2.16, according to the gnupg ticket: https://dev.gnupg.org/T5931#157711

I guess this would require a firmware > RTM.12?

szszszsz, 2022-05-06:
I have just confirmed that there are no 1.2.16 commits in the latest Nitrokey Start firmware release (so it is a plain 1.2.15, not 1.2.15+).

dvzrv, 2022-05-22:
@szszszsz Are there any plans of a firmware upgrade, which makes use of gnuk >= 1.2.16?

Currently this setup limits me to using openssh 8.8p1 on all clients and the new KexAlgorithms default `sntrup761x25519-sha512@openssh.com` has been put in place to prevent "record now, decrypt later" type of attacks.

szszszsz, 2022-05-25:
Hi! I am sorry, but the tests results got delayed. Perhaps we can make the release with 1.2.16+ without any further waiting. I will update this ticket about that.

seifertm, 2022-06-26:
I'm also affected by this issue for some time now. Is there something I can do to support this?

szszszsz, 2022-07-01:
Hi! 
Just a quick update: we have work on the new release scheduled for the next week, which should fix this ticket.

seifertm, 2022-10-24:
I installed RTM.13-RC2 today and I can confirm it fixes the issue.

szszszsz, 2022-10-24:
Great to hear that! If that's not too much to ask, can you test other features too? E.g. multiple identities?

dvzrv, 2022-11-19:
Hi, any movement on this?

dvzrv, 2022-11-19:
I've now tested this with two identities with firmware RTM.13-RC2 and that seems to be good with both (and client side openssh >= 9.1p1

seifertm, 2022-11-21:
Thank you. I didn't test with multiple identities, but I've been using the new firmware for a month without issues.

szszszsz, 2022-12-08:
@dvzrv I plan it today. I've done the tests yesterday, and all looks good.

closed 2022-12-08 by szszszsz